gonzalo/gphoto2-updater #14: Make tag for each gphoto version

Sija, 2015-01-19:
It would be much helpful when trying to install given gphoto version. Then we could use `https://raw.githubusercontent.com/gonzalo/gphoto2-updater/[v2.5.5]/gphoto2-updater.sh` url form. Otherwise we need to resort to using commit SHA.

Thanks for consideration and for such a helpful project :)

gonzalo, 2015-01-19:
That sounds cool! I have never used before git tags. I'll give it a try in future developments.

Thanks for your feedback!

Sija, 2015-01-20:
No problem, this script saved me some time during development and that motivates :) For tag/release management you can use fine [github web ui](https://github.com/gonzalo/gphoto2-updater/releases/new), retroactively too (`v2.5.5` plz?) :)

gonzalo, 2015-01-21:
Nice! yesteday I was looking about tags but I get stuck about how to do it from web UI. Let me take a look to do it

gonzalo, 2015-01-21:
Here you will find 2.5.5 version https://github.com/gonzalo/gphoto2-updater/releases/tag/2.5.5

...and obviously lastest version is https://github.com/gonzalo/gphoto2-updater/releases/latest

Thank you for introducing me to tags and releases in GIT

Sija, 2015-01-21:
Swell news, thanks! :)

closed 2015-01-21 by Sija