gonzalo/gphoto2-updater #16: libgphoto2 >= 2.5.6 not met. Found: 2.4.14

Xophe92, 2015-02-21:
Hi, 

during the ./configure of gphoto2 I guess, I obtain a message:

checking if libgphoto2 version is matching requirement >= 2.5.6... no
configure: error: Version requirement libgphoto2 >= 2.5.6 not met. Found: 2.4.14

I deleted "/usr/local/lib/libgphoto*" and started again and made sure that nothing was linked to gphoto in /usr/lib...

Please advice on the way I could delete this 2.4.14 on my system :)

Regards,

Christophe

gonzalo, 2015-04-21:
"sudo apt-get remove libgphoto2" is probably the best way to do it

gonzalo, 2015-04-21:
Also I recommend to download new gphoto2-updater version wich fix a 404 error when trying to dowlad libusbx_1.0.17

closed 2015-04-21 by gonzalo