gonzalo/gphoto2-updater #18: Unable to get libusbx_1.0.17

benoitguigal, 2015-04-02:
http://ftp.de.debian.org/debian/pool/main/libu/libusbx/libusbx_1.0.17.orig.tar.bz2 => 404

frxnz, 2015-04-02:
I just happen to be having the same issue right now. I tracked down the package and the Debian site and made [the change here](https://github.com/frxnz/gphoto2-updater/blob/libusbx-patch/gphoto2-updater.sh#L78)

I then pointed the install command to my fork:
```
$ wget https://raw.githubusercontent.com/frxnz/gphoto2-updater/libusbx-patch/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh
```

The install seems to be going well now.

pepade84, 2015-04-15:
or actually I could use the lastest version from sourceforge...

if wget -q http://downloads.sourceforge.net/project/libusbx/releases/1.0.18/source/libusbx-1.0.18.tar.bz2
	then
		tar xjvf libusbx_1.0.18.orig.tar.bz2
		cd libusbx-1.0.18/
	else
		echo "Unable to get libusbx_1.0.18"
		echo "Cleaning and exiting..."
		exit 1
fi

echo 
echo "--------------------------------------"
echo "Compiling and installing libusb 1.0.18"
echo "--------------------------------------"

gonzalo, 2015-04-21:
Fixed using an alternative repository.

closed 2015-04-21 by gonzalo

mxjones, 2015-07-04:
I had this issue today, fixed it using frxnz version 

http://snapshot.debian.org/archive/debian/20130909T094354Z/pool/main/libu/libusbx/libusbx_1.0.17.orig.tar.bz2

gonzalo, 2015-07-10:
fixed in new version using http://downloads.sourceforge.net/project/libusbx/releases/1.0.18/source/libusbx-1.0.18.tar.bz2