gonzalo/gphoto2-updater #28: get same segument falt problem after update script

Veallym0n, 2015-09-28:
Hi. I got the same segument fault problem like 
https://github.com/lwille/node-gphoto2/issues/54

and I tried to update with your script. seems it not working.
can you helpme?

my current version of libgphoto2 is 

```
root@raspberrypi:~/astrock# gphoto2-config --version
libgphoto2 2.5.8
```

and i got the segument fault error like this.

```
root@raspberrypi:~/astrock# cat test.js
var gphoto2 = require('gphoto2')

camera = undefined
var GPhoto = new gphoto2.GPhoto2();

console.log('[trying to get Camera]')
GPhoto.list(function(list){
    if (list.length === 0) return;
    camera = list[0];
    console.log(JSON.stringify(camera)+'\n')
    console.log('[trying to get Settings]')
    camera.getConfig(function(er,sett){
        console.log(sett)
    })
})

root@raspberrypi:~/astrock# node test.js
[trying to get Camera]
{"model":"USB PTP Class Camera","port":"usb:001,009","_gphoto2_ref_obj_":{}}

[trying to get Settings]
Segmentation fault
```

at gdb it's

```
root@raspberrypi:~/astrock# gdb node
Reading symbols from /usr/local/bin/node...done.
(gdb) run test.js
Starting program: /usr/local/bin/node test.js
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb6cea460 (LWP 12128)]
[trying to get Camera]
[New Thread 0xb6ba4460 (LWP 12194)]
[New Thread 0xb63a4460 (LWP 12196)]
[New Thread 0xb5ba4460 (LWP 12199)]
[New Thread 0xb53a4460 (LWP 12200)]
{"model":"USB PTP Class Camera","port":"usb:001,009","_gphoto2_ref_obj_":{}}

[trying to get Settings]

Program received signal SIGSEGV, Segmentation fault.
0x00000000 in ?? ()
(gdb) bt
#0  0x00000000 in ?? ()
#1  0xb6ccb440 in open_camera(_Camera**, std::string, std::string, _GPPortInfoList*, _CameraAbilitiesList*) ()
   from /home/kevin/astrock/node_modules/gphoto2/build/Release/gphoto2.node
(gdb)
```

is that version wrong?

gonzalo, 2015-09-28:
This a question for libgphoto2 developers. Recently they have moved to GitHub on https://github.com/gphoto. You should ask to them.

closed 2015-09-28 by gonzalo

Veallym0n, 2015-09-29:
ok. sorry. 
thanks alot