gonzalo/gphoto2-updater #30: raspbian Jessie stuck on libgphoto2 2.5.4

Benjamin7785, 2015-11-25:
Hey,
running gphoto2-updater on a vanilla raspbian jessie Image seems to fail updating the libgphoto2. while everything updates fine only libgphoto2 remains Version 2.5.4. Any idea to fix this is appreciated.
Cheers Ben

P.S.: any chance telling when gphoto2-updater 2.5.9 will be available :-)

gonzalo, 2015-11-27:
Very strange. This script specifically downloads and install version 2.5.8. Maybe you have manually installed (via apt-get) other version. Please be sure to remove it via apt-get remove.

Anyway GitHub-integration branch is your option if you want to install the last gphoto2 version. Take a look in https://raw.githubusercontent.com/gonzalo/gphoto2-updater/GitHub-integration/gphoto2-updater.sh

gonzalo, 2015-11-27:
Master branch now takes 2.5.9 version from original repositories. Please take a look and check if it works for you.

Benjamin7785, 2015-11-29:
Hey,
Unfortunately, there is no success so far. Even trying to update to 2.5.9 leaves libgphoto2 at version 2.5.4. Whereas gphoto2 updates to 2.5.9 nicely. 
There is a youtube video online showing exactly the same problem I have.
https://www.youtube.com/watch?v=hQ14nB6zhcc

scribblemaniac, 2015-11-29:
Based on that video, I would guess that gphoto2 is not in your `PATH` variable. What is the output when you run `whereis gphoto2`? What is the output when you run `echo $PATH`?

gonzalo, 2015-11-30:
As @scribblemaniac said it should be something related to PATHs. Problably, your OS is calling the apt-get version of libgphoto2 instead of the newer one. Be sure that you have uninstalled gphoto2 and libgphoto2-2 packages before executing the script.

gonzalo, 2015-12-02:
Confirmed by Andr.oid Eric, the problem seems to be the the libgphoto2-port10 package that comes preinstalled on debian jessie. Be sure to remove it manually before executing script

    sudo apt-get remove libgphoto2-port10

http://helloraspberrypi.blogspot.com.es/2015/12/install-gphoto2-libgphoto2-version-259.html

scribblemaniac, 2015-12-02:
Should we consider adding this line to the script?

Benjamin7785, 2015-12-02:
Okay, removing libgphoto2-port10 did indeed work perfectly. Thanks for the support.

closed 2015-12-02 by Benjamin7785

gonzalo, 2015-12-03:
Instruction for removing package libgphoto2-port10 will be included in new script version version.