gonzalo/gphoto2-updater #31: Raspberry Pi Zero

scribblemaniac, 2015-12-02:
As you may already know, the Raspberry Pi Foundation recently released the [Raspberry Pi Zero](http://swag.raspberrypi.org/products/pi-zero). As far as I know, this supports gphoto2 just as well as the other boards. The issue is that they don't have networking capabilities so this current updater will not work for most people. Is this a challenge that we want to address or should we just say it's not supported?

I'm pretty sure this could be implemented by downloading gphoto2 and its dependencies onto external storage (sd, usb, etc.) from a computer with internet. The storage device could then be connected to the Pi Zero and the packages could be built and installed from there. There are obviously going to be some additional complexities for the script it we go this way though.

gonzalo, 2015-12-02:
Probably the "easiest" way to do this is generate both packages gphoto2 and libghoto2, check other dependencies and offer a collection of files to install with dpkg. So, it could be interesting to change this project to a debian packages generator. This would make easy to revert changes on system if at any moment we want to uninstall the software. We consider that packgages should be available for different architectures as i386, amd64 and armh.

We could use this thread to collect resources about how to create that packages and list other packages needed to install at a fresh debian/raspbian/ubuntu distribution.

scribblemaniac, 2015-12-02:
An interesting idea. I don't have any experience making debian packages, but I think this could work well. I'll start things off:
[Introductory tutorial slides](https://www.debian.org/doc/devel-manuals#packaging-tutorial)
[Developer reference (advanced)](https://www.debian.org/doc/manuals/developers-reference/)

I'll go over the first one at the very least and post here again when I get an idea of what needs to be done.

scribblemaniac, 2015-12-09:
Okay, I've done a bit of research and this is the script workflow that I propose:
1. Download packages for dependencies (for the target architecture).
2. Download a 'template' source package from this repo that contains most of the settings for gphoto2 and libgphoto2.
3. Modify the source package as necessary (add in latest code from github for example)
4. At this point, the execution can be stopped and all of the files can be transferred to the target system. A connection to the internet is no longer required.
5. On the target device, build the modified source packages for libgphoto and gphoto2.
6. Now install all packages.

Of course if you are installing on a device with internet, step 4 can be skipped. Unfortunately, we can't just provide a .deb file like most packages because we support the latest version. The package would have to be rebuilt and reuploaded anytime a change was made to the github repositories, or it would quickly become out of date.