gonzalo/gphoto2-updater #34: camlibs/ptp2/library.c file not found

Benjamin7785, 2015-12-29:
Hi,
Due to a bug inside libgphoto2 I need to manually edit a file. But I cannot find the file that is mentioned here https://github.com/gphoto/libgphoto2/issues/13
I wonder if that is because I used your script to update gphoto2 on my Raspbian Jessie.
Sorry, I am just not an expert in this.... :-)
Thanks, Ben

scribblemaniac, 2015-12-29:
This bug appears to have been patched already. I would recommend reinstalling gphoto2 and libgphoto2. The issue has apparently been patched in 2.5.9, so if you are still having issues, it is likely a new issue with libgphoto2.

The file you are looking for is likely not on your computer. The file you are looking for is a source file, however what gets installed is only the compiled executable files. If you want to edit the file you are looking for, you will have to clone the libgphoto2 directory from github, modify your file, and then install it manually.

Benjamin7785, 2015-12-29:
thanks for the support
Seems like I need a bit more investigation into this issue. I thought I was using 2.5.9 already....
Am I right that with deinstall you mean "apt-get remove gphoto2" ?
But how do I remove libgphoto2? Can you give me a hint?

scribblemaniac, 2015-12-29:
If you rerun our script, it *should* clear out old version and install the latest (make sure you have the latest version of our script). If you want to install it manually, then you can try `apt-get remove libgphoto2-6`. It also might be helpful to run `gphoto2 --version` first. At the beginning of the output it will list the version of gphoto2, but if you look down a bit it will list the version of libgphoto2, which is what is important in this case.

Benjamin7785, 2015-12-29:
Just wondering: when new code has been added to gphoto on git, will running gphoto-updater.sh install the new code directly? As far as I understood the script is downloading from github/gphoto directly, isn't it?

scribblemaniac, 2015-12-29:
Yes, it will install the latest code on github by default.

gonzalo, 2016-01-04:
that's right. Installer will ask if you want to install last stable version or development version

closed 2016-03-21 by gonzalo