gonzalo/gphoto2-updater #35: Problems with sh file

raffaelecolleo, 2016-01-04:
Hi Gonzalo,
I'm having problems installing the libraries on a raspberri pi 2.
I have the lite version of jessie installed and cannot install libgphoto2. 
line 141: git: command not found.

There is also a problem on libusb 1.0.20
tar: Child returned status 2
tar: Error is not recoverable: exiting now.

Could You please check this errors

Thanks a lot

gonzalo, 2016-01-04:
That's very strange because git package should be installed at the beginning of script.

Please check that you have git installed (git -v) and if not install it manually with "sudo apt-get install git" and try again.

gonzalo, 2016-01-22:
Hi @raffaelecolleo ! any updates about this issue?

gonzalo, 2016-02-11:
Closed by inactivity

closed 2016-02-11 by gonzalo