gonzalo/gphoto2-updater #36: libusb 1 confusion

msmeissn, 2016-02-10:
can you perhaps try to use the system libusb 1 ...? 

building a different libusb 1 version than the one on system, but using them causes reports where
libusb_strerror is missing

libusb-1.0-0-dev

is the ubuntu packagename for the development files I think.

gonzalo, 2016-03-23:
Hi @msmeissn , not sure why I'm using libusb 1.0.20, probably it was a kind of requierement when I began this project.

I'll give a try to system library libusb-dev and check if libgphoto2 and ghoto2 compile right

gonzalo, 2016-04-06:
After testing it in Debian Jeezy / Wheezy, Ubuntu 14.04 and Raspian 7 / 8, use of system libusb dev package seems to work correctly. Compilation works fine.

Change pushed to the master branch

closed 2016-04-06 by gonzalo