gonzalo/gphoto2-updater #40: Rollback to a previous version?

Nic2012, 2016-04-03:
How can I use the script to install one of the previous versions e.g. 2.5.7 ?

Thank y for y effort providing us this batch :)
Best regards,
Nic

gonzalo, 2016-04-05:
Check different releases at https://github.com/gonzalo/gphoto2-updater/releases

Nic2012, 2016-04-05:
Hmmh, its a bit confusing, each paragraph of previous version describes same link to the current update script:
..
Gphoto2 and libgphoto2 2.5.7 compiler and installer script
http://github.com/gonzalo/gphoto2-updater...
..
You probably meant the part under downloads?

scribblemaniac, 2016-04-05:
Yes, it's the downloads you want, not that link.

gonzalo, 2016-04-05:
Each realease attaches below paragraph the source code of tht release

Nic2012, 2016-04-05:
BTW the new 2.5.10 is out, according to release notes there is bugfix that could help me. That was the reason why I wanted to rollback. Do y have plans to upgrade the updater :)

gonzalo, 2016-04-06:
By default, the master branch offers the possibility of installing lastest development version (at this moment 2.5.10.1). 

Anyway soon I will publish a new version including lastets stable version 2.5.10.

gonzalo, 2016-04-06:
New version of script for 2.5.10 is available

Nic2012, 2016-04-07:
Great!
Update to 2.5.10 works fine. Thank y very much.

closed 2016-04-07 by Nic2012