gonzalo/gphoto2-updater #44: bash: /usr/bin/gphoto2: No such file or directory

mxa, 2016-08-15:
After running the script, it seems I no longer have gphoto2.
Installer script output:
https://bpaste.net/show/0b7694d34a0b

scribblemaniac, 2016-08-15:
Thanks for reporting this issue!

@gonzalo Seems like the `core` variable initialization accidentally got deleted [here](https://github.com/gonzalo/gphoto2-updater/commit/4a19d5cdf3b984fd120d18d6b2458a325b01fac1#diff-289783193eb5c5751103cc9cd88c37deL176), and I think that is what is causing this issue. I can't push to github at the moment so could you please add this line back in?

mxa, 2016-08-16:
I confirm this does the trick.

gonzalo, 2016-09-01:
Thanks for your report and commit.

I've been "out of business" during holidays. Give me a couple of days to test it.

gonzalo, 2017-01-12:
Commit accepted

closed 2017-01-12 by gonzalo