gonzalo/gphoto2-updater #46: Install libgphoto2 to system (?)

dcholth, 2016-12-24:
I used your fantastic install and update script on my RaspberryPI.  However, I'm experiencing the issue [documented here](http://gphoto-software.10949.n7.nabble.com/2x-Sony-Alpha-5100-set-config-td16208.html) where I can't set my configurations.

The suggested fix is to install libgphoto2 into system, and not /usr/local.  When I look through my file system after running the script, it appears that libgphoto2 was installed in /usr/local.  Is there a way that the script can install it into system?

Sorry, I'm pretty green when it comes to linux and a lot of this stuff, so this my issue may be bogus, but its the only end I've reached in my research.

scribblemaniac, 2016-12-24:
This script does install to /usr/local, but I'm a bit skeptical if that is the real problem. What is the output of `gphoto2 -v`?

gonzalo, 2016-12-26:
Did did you have gphoto installed before using our script? Maybe system is
calling the official installation ( you can check it with command "which
gphoto2"

Be sure to uninstall gphoto2 and libgphoto2 with apt-get remove and then
install last version with the script

dcholth, 2017-01-06:
Thanks for the response!  I ended up wiping my Pi and did the install using the very latest release option, and it appears to have fixed the problem.  Appreciate the response!  Sorry I couldn't get you more useful information on why it occurred in the first place.

closed 2017-01-12 by gonzalo