gonzalo/gphoto2-updater #48: Trouble updating

uzelessknowledge, 2017-04-27:
Hi. I've been having trouble getting my version updated to support a camera that was just added in .12 and .13. Running Raspberry Pi model B with Jessie/Pixel. 
Ran the script on the pi tonight and still can't get libgphoto2 to update.
Even tried starting with a fresh Pixel install on the SD.
Ok this is strange, but worked. I thought about trying to remove libgphoto2 after running the script and suddenly gphoto2 --version shows libgphoto2 as 2.5.12 instead of the 2.5.4 it was showing. 
Gphoto2 --auto-detect shows the camera now.

I ran sudo apt-get remove libgphoto2*.
Maybe something needs to be updated in the install process?
I'd still like to get on .13, but after trying for almost 8 hours to get this far I'm happier. 
Now to see if I can get my Ubuntu working. 
[image]

scribblemaniac, 2017-04-27:
Did you select development or stable? If you select stable, it will install .12 since we haven't updated the script since the release of that version. Now that this has come to my attention, I'll make a new PR shortly to update it. In the mean time, you should be able to use the development version.

scribblemaniac, 2017-04-27:
Okay, I've pushed an update to [here](https://github.com/scribblemaniac/gphoto2-updater/tree/update-2-5-13). You can use that if you want, or wait until #49 gets merged.

uzelessknowledge, 2017-04-27:
Thank you for your help. I saw your script has apt-get remove -y gphoto2 libgphoto2*, so this should've removed the instance of 2.5.4, right. Or does that need to be run as sudo to do the remove?
I don't know and I'm just asking since I've had a struggle even manually getting updated.

uzelessknowledge, 2017-04-27:
I have not had the opportunity to try your new script. Didn't know how to use it without using the wget from the instructions.
I was able to manually install, but still had to run the remove for the correct version to show.
I needed some dependencies. Took a bit to figure those out. Here is what I did on a fresh image.
1. Install libltdl-dev libusb-dev libusb-1.0 libexif-dev libpopt-dev
2. Install lbgphoto2 from the source. v.2.5.13
3. Install Gphoto2 from source v2.5.11
4. Check Gphoto version - version is 2. 5.11, but libgphoto is 2.5.4
5. sudo apt-get remove libgphoto2*
6. Check version again and this time Gphoto is 2.5.11 and libgphoto is 2.5.13

scribblemaniac, 2017-04-27:
That sounds like more or less like what the script does. The script also sets up some important udev rules and stuff like that, but since you probably didn't delete those, the should still be working for your manual installation. The updater script is always run as root, so you shouldn't have to worry about apt-get remove or anything like that. Sounds like you got it working, but if you feel like temping fate, here's a one-liner for the 2.5.13 version:
```bash
wget https://raw.githubusercontent.com/scribblemaniac/gphoto2-updater/update-2-5-13/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh
```

uzelessknowledge, 2017-04-28:
That works to update without me having to do a manual remove. It works on the pi in Jessie to use Gphoto2 to trigger the camera. Thanks.

However, I'm also trying to get Stars/Ekos working in Ubuntu and it updates the libgphoto2 to 2.5.13 as expected, but it's also removing files from /usr/bin that Ekos uses. Namely INDI_CANON_CCD and INDI_GPHOTO_CCD. This is at least what I can tell. I have not had a chance to work with the INDI files in Jessie. That is another configuration I want to see if works for me. I'm assuming the same is going to happen.

This link does not remove those files. This link is loading 2.5.12, so I don't know if the problem is in 13b or in this script.
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh

I'm working both Ubuntu and Jessie to see what will work bette for what I want.

gonzalo, 2017-04-28:
Script updated to 2.5.13.

uzelessknowledge, 2017-04-28:
Gonzalo does that mean it's update based on my last comment about deleting files for Ekos or just updated for Jessie?

scribblemaniac, 2017-04-29:
If you had trouble with my script, the update will not help you. I have no experience with indi files, and am a bit surprised they were even there to begin with. You can perhaps try installing this package:
http://indilib.org/devices/ccds/gphoto.html

uzelessknowledge, 2017-04-29:
Hey and thank you for all the help. If I were just relying on gphoto the. Your script worked great. It let me install on a fresh image and control the 80D. 
BUT since I want to use indi I'm having issues. The install of indi has 2.5.4 and of course that doesn't work with the 80D. Seems I can't get them both installed at the same time.

scribblemaniac, 2017-04-29:
The following seemed to work for me. First install the stable version of gphoto2 with our installer script. Then run:
```
# Install dependencies
sudo apt-get install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libraw-dev
# Install libindi
git clone https://github.com/indilib/indi.git
cd indi
mkdir -p build/libindi
cd build/libindi
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../libindi
make
sudo make install
# Install indi gphoto driver
cd ../
mkdir indi-gphoto
cd indi-eqmod
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release ../../3rdparty/indi-gphoto
make
sudo make install
```
I'm left with libgphoto2 2.5.13, gphoto2 2.5.11, and the cannon and gphoto indi files in /usr/bin. If you are still having difficulties and want help here, state the exact steps you are taking, and at which step the indi files are getting deleted.

uzelessknowledge, 2017-04-30:
Thank you for the help. Here are my results. I got an error and stopped there.
Fresh install of Jessie
Didn’t change any settings in the image. Usually I set up locale and such.

sudo apt-get update
#used your script except I changed the variable to to 2.5.12 since .13 doesn’t seem to like the Canon 80D. I edited the filed before running it. Downloaded with this line, then edited, and then ran it.
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh
gphoto2 —version
#gphoto2 2.5.11
#libgphoto2 2.5.12
#confirmed gphoto2 works with Canon 80D by capturing an image
# Install dependencies
sudo apt-get install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libraw-dev
#E:unable to locate package libgsl-dev. Removed the from the line and ran the rest
# Install libindi
git clone https://github.com/indilib/indi.git
cd indi
mkdir -p build/libindi
cd build/libindi
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../libindi
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find GSL (missing: GSL_INCLUDE_DIR GSL_LIBRARY GSL_CBLAS_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /home/pi/indi/cmake_modules/FindGSL.cmake:158 (find_package_handle_standard_args)
  CMakeLists.txt:237 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/indi/build/libindi/CMakeFiles/CMakeOutput.log".
See also "/home/pi/indi/build/libindi/CMakeFiles/CMakeError.log".

[Archive.zip](https://github.com/gonzalo/gphoto2-updater/files/966533/Archive.zip)

scribblemaniac, 2017-04-30:
The build fails because apt couldn't find libgsl-dev. Maybe it's under `libgsl0-dev` or something like that so just do `apt search libgsl` and see if you can figure out which one you need. Once you've got it installed, redo the libindi install.

uzelessknowledge, 2017-04-30:
I saw that but wasn't sure if that was right so I wanted to wait for your guidance.

#Installed libgsl0-dev
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../libindi
make
sudo make install

#Install indi gphoto driver
cd ../
mkdir indi-gphoto
**cd indi-eqmod
#not sure on this because there is no directory indi-eqmod in build. I did cd into indi_gphoto
cd indi-gphoto**
make
sudo make install

It works!!! THANK YOU SO MUCH!!! I can start the indi server on the RPi-Jessie and then connect to it from the client Kstars on the Mac. Now to figure out how to use it. 

I assume this same process should work on Ubuntu? I can't use their package because, like here, it installs libgphoto2 using the old version.

I truly appreciate your help!

scribblemaniac, 2017-04-30:
> cd indi-eqmod

Oops, just a typo, but it looks like you figured it out. I'm glad I could help! 😁 The same process should work on Ubuntu, or any debian-based distro. You may have to search for the names of a couple of the packages like you did for libgsl this time.

uzelessknowledge, 2017-04-30:
Success thanks to you! Thank you again.
I've written up a doc to share on the Indi forum.
[Indi Server Raspberry Pi.pdf](https://github.com/gonzalo/gphoto2-updater/files/967047/Indi.Server.Raspberry.Pi.pdf)

closed 2017-05-09 by uzelessknowledge