gonzalo/gphoto2-updater #51: Permission problem for non-root user

solfolango, 2017-05-18:
This is on Raspbian Jessie Lite:

I am running into a permission error stating that I cannot claim the USB device. Running with sudo works fine.

```
gphoto2 v2.5.11
libgphoto2 2.5.4
libgphoto2_port 0.12.0
```

lsusb `ID: 04a9:30ee, Canon, Inc. EOS 350D`

I then reinstalled gphoto2 from apt-get, there I can run gphoto2 with the user pi (regular user). I checked and it created a new rules-file and saved the existing one with a .save extension. Running a diff on both files just states one line with a possible typo: Instead of `GROUP="plugdev"` in the .save file it says `GUP=plugdev"` on one single line. I can't tell if this one is the culprid.

gonzalo, 2017-05-19:
have you tried to reinstall last versions of gphoto2 with our script?

solfolango, 2017-06-04:
Sorry to let you hang for so long, but my days are quite busy. 

I ran your script on a clean raspbian install on an raspberry pi 3. It then only worked with sudo. Since I access gphoto from a nodejs application, I didn't want to run it as sudo, therefore I tried the "official", not up-to-date version that is in the repositories, that one worked fine. By comparing the rules-file, I noticed the difference I mentioned above.

I haven't since ran either your script nor an "apt-get install gphoto2", but I can try it again if my time permits.

closed 2018-01-08 by gonzalo