gonzalo/gphoto2-updater #54: udevd: command not found on raspian 4.9.53

rock-meister, 2017-10-10:
running on rpi3 - raspberrypi 4.9.53

The script errors out at this spot
---------------------------------------------------------------------------------
Generating udev rules, see http://www.gphoto.org/doc/manual/permissions-usb.html
---------------------------------------------------------------------------------

./gphoto2-updater.sh: line 228: udevd: command not found
./gphoto2-updater.sh: line 230: [: : integer expression expected
./gphoto2-updater.sh: line 233: [: : integer expression expected
./gphoto2-updater.sh: line 236: [: : integer expression expected

I checked the path and packages...
pi@raspberrypi:~ $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

pi@raspberrypi:~ $ apt list --installed|grep udev
libgudev-1.0-0/stable,now 230-3 armhf [installed,automatic]
libudev-dev/stable,now 232-25+deb9u1 armhf [installed]
libudev0/stable,now 175-7.2 armhf [installed]
libudev1/stable,now 232-25+deb9u1 armhf [installed]
udev/stable,now 232-25+deb9u1 armhf [installed]
pi@raspberrypi:~ $

scribblemaniac, 2017-10-10:
I dug through the man pages on Ubuntu and it uses `/lib/systemd/systemd-udevd --version` instead of `udevd --version`. I can't verify if this will work on the pi though.

rock-meister, 2017-10-10:
Both  /sbin/udevadm --version and /lib/systemd/systemd-udevd --version returns 232.  Not sure if Ubuntu has /sbin/udevadm...  if so, we can substitute udevd with udevadm to get the version number.

Anyway, I changed udevd to udevadm and was able to generate /etc/udev/hwdb.d/20-gphoto.hwdb successfully.
Thanks a bunch

closed 2017-10-10 by rock-meister