gonzalo/gphoto2-updater #57: This script removes ubuntu-desktop!

BeebBenjamin, 2018-01-03:
This script removes ubuntu desktop and other software:

Removing colord (1.0.6-1) ...
Removing deja-dup-backend-gvfs (30.0-0ubuntu4.1) ...
Removing ubuntu-desktop (1.325) ...
Removing software-center (13.10-0ubuntu4.2) ...
Removing gvfs-backends (1.20.3-0ubuntu1.2) ...
Removing printer-driver-postscript-hp (3.14.3-0ubuntu3.4) ...
Removing hplip (3.14.3-0ubuntu3.4) ...
Removing libgphoto2-2-dev (2.5.3.1-1ubuntu2.2) ...
Removing shotwell (0.18.0-0ubuntu4.5) ...
Removing simple-scan (3.12.3-0ubuntu1) ...
Removing libgphoto2-dev (2.5.3.1-1ubuntu2.2) ...
Removing libgphoto2-dev-doc (2.5.3.1-1ubuntu2.2) ...
Removing libgphoto2-l10n (2.5.3.1-1ubuntu2.2) ...
Removing wine (1:1.6.2-0ubuntu4) ...
Removing sane-utils (1.0.23-3ubuntu3.1) ...
saned disabled; edit /etc/default/saned
Removing libsane:amd64 (1.0.23-3ubuntu3.1) ...
Removing wine1.6-amd64 (1:1.6.2-0ubuntu4) ...
Removing libgphoto2-6:amd64 (2.5.3.1-1ubuntu2.2) ...
Removing wine1.6 (1:1.6.2-0ubuntu4) ...
Removing libgphoto2-port10:amd64 (2.5.3.1-1ubuntu2.2) ...
Removing wine1.6-i386 (1:1.6.2-0ubuntu4) ...

The command "sudo apt-get remove libgphoto2* is the culprit.

Very sloppy.

gonzalo, 2018-01-08:
Thanks for the feedback. Obviously, removing ligphoto2 which is a side library, should not remove the complete ubuntu desktop. Can you confirm wich version of ubuntu are  you using?

BeebBenjamin, 2018-01-08:
Ubuntu 14.10 LTS.  It shouldn't but it does.

gonzalo, 2019-01-08:
Finally I've removed the gphoto and libghoto uninstall and just override libraries if they are previously installed.

closed 2019-01-08 by gonzalo