gonzalo/gphoto2-updater #58: Capture not supported by SONY UMC-R10C

sameenashaikh29, 2018-02-16:
Hi,
I installed the gphoto2 library using the sudo apt-get install gphoto2 command. My camera was not getting auto detected. I then updated the library through https://github.com/gonzalo/gphoto2-updater/ and the camera got detetcted using the gphoto2 --auto-detect command. It is listed ined in the supported cameras list on gphoto2 on site and sudo --list-cameras too gave me this camera in the list. But capture-image and --trigger-capture command doesnt work.. Ive listed outputs of some commands below.
Any help appreciated

pi@raspberrypi:~ $ lsusb
Bus 001 Device 006: ID 413c:2107 Dell Computer Corp. 
Bus 001 Device 005: ID 0458:003a KYE Systems Corp. (Mouse Systems) NetScroll+ Mini Traveler / Genius NetScroll 120
Bus 001 Device 008: ID 054c:0a79 Sony Corp. 
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@raspberrypi:~ $ gphoto2 --version
gphoto2 2.5.15

Copyright (c) 2000-2017 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.15         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

pi@raspberrypi:~ $ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony UMC-R10C                  usb:001,008     

pi@raspberrypi:~ $ gphoto2 --abilities
Abilities for camera             : Sony UMC-R10C                               
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Capture not supported by the driver
Configuration support            : no
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes

pi@raspberrypi:~ $ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: UMC-R10C
  Version: 2.00
  Serial Number: 00000000000000004205006000000079
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:

Device Property Summary:

pi@raspberrypi:~ $ gphoto2 --capture-image
                                                                               
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

scribblemaniac, 2018-02-16:
You have the right version of gphoto2 installed. If you look at the gphoto2's [list of supported cameras](https://web.archive.org/web/20180128011949/http://gphoto.org/proj/libgphoto2/support.php), the Sony UMC-R10C is there, but that just means that downloading images from that camera is supported. It is not on the [list of cameras that support remote control](https://web.archive.org/web/20180113232926/http://gphoto.org/doc/remote/). You will have to work with the upstream project libgphoto2, not this updater, to request remote control support for your camera (which I see you've [already done](https://github.com/gphoto/libgphoto2/issues/238)).

sameenashaikh29, 2018-02-17:
Hi,
Thank you for the response. I'll follow up with  that project..

Thanks & Regards
Sameena Shaikh

closed 2018-02-19 by gonzalo

cfort38, 2022-03-30:
Hi
I just bought a Sony UMC R10C. Did any of you manage to make it work with DROTAG ?
Thank you
Regards
Charles

gonzalo, 2022-03-30:
You'll probably have better luck asking in libphoto2 forums https://github.com/gphoto/libgphoto2/issues/238
This repository is only for the upgrade script.

cfort38, 2022-03-30:
Thank you !

Regards

charles