gonzalo/gphoto2-updater #64: git fails for libgphoto2 and gphoto2

FormerLurker, 2018-08-08:
Very nice script, by the way!  So much easier than doing all this manually.  Thank you for your efforts!

When I ran the script I got an error about trying to run git with sudo.  I had to change 

'''
if (git clone  $branch_gphoto https://github.com/gphoto/gphoto2.git)
'''
to
'''
if (/usr/bin/git clone  $branch_gphoto https://github.com/gphoto/gphoto2.git)
'''
for both gphoto2 and libgphoto2.

I'm still learning about linux, so hopefully my fix wasn't a big no-no.  Hopefully this will help you or someone else.

gonzalo, 2018-08-17:
thanks @FormerLurker for your feedback but it seems related to your particular configuration. GIT should be available from your path. Maybe you have more than one git installed?

FormerLurker, 2018-09-10:
Thanks for responding.  I thought I replied to this earlier, but it looks like I did not!  Feel free to close this ticket out since I now know there is some trickery going on in OctoPi that makes my changes necessary, but only for that particular distribution.

Maybe I can ask a quick noob question?  Is there a quick way to uninstall gphoto2 after running this script?  `apt list `doesn't show gphoto2, but `compgen -c` does.  I want to uninstall it because I'm creating a tutorial and want to show the installation of gphoto2.  If all else fails I can just re-flash my sd card, but I'd rather not :)

closed 2018-09-27 by gonzalo