gonzalo/gphoto2-updater #69: Tried the script and got version 2.5.15

Creat1veM1nd, 2019-01-26:
Hello,

I wanted to install gphoto2 via your script. It basically ran through, but in the end, I got this: 

`gphoto2 2.5.15

Copyright (C) 2000-2017 Lutz Müller und andere

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

Diese Version von gphoto2 benutzt die folgenden Softwareversionen und Optionen:
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
`

I expected version 2.5.22, as stated in the script. 
Did I something wrong ?

Thanks in advance,

Creat1veM1nd, 2019-01-27:
I tried it on another PC and there all worked well. So I guess the problems was somewhere deep inside the configuration of the PC itself.

closed 2019-01-27 by Creat1veM1nd

scribblemaniac, 2019-01-28:
@Creat1veM1nd Your issue is probably because you have another version of gphoto2 installed (ex. from apt). This script used to run `apt-get remove gphoto2 libgphoto2*`, but does not anymore because it caused the removal of important packages for some users. You can try and use this command to fix your issues, but just check to make sure it doesn't remove something like `ubuntu-desktop`. It will ask for a confirmation first before doing anything.

Creat1veM1nd, 2019-01-29:
I removed the old version via apt before using the autoupdater but got the same result. Probably something was wrong while removing, at least at this one PC (Linux in a VM on a Windows10 Host) - on a real Linux machine it worked like a charm.