gonzalo/gphoto2-updater #79: Running ghoto2-updater on rpi does not result in latest version 2.5.26 but 2.5.23.1

Nears, 2020-12-27:
Running ghoto2-updater on rpi does not result in latest version 2.5.26 but 2.5.23.1.

What could be the reason for this? Am I missing any preconditions which are not mentioned in the How-To?

scribblemaniac, 2020-12-27:
Likely you have that version installed with a package manager. Try running `sudo apt-get remove gphoto2 libgphoto2*`, but make sure that it doesn't remove anything else important as a result. If it does remove something like gnome-desktop, then you may be able to make this script work by applying the changes from #73 instead.

Nears, 2020-12-27:
Thanks for the quick feedback, this solved it partially. Now I have the lib at the latest version, but not yet gphoto2

See console log:

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.26         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

scribblemaniac, 2020-12-27:
What is the output of `which gphoto2`?

Nears, 2020-12-27:
/usr/local/bin/gphoto2

scribblemaniac, 2020-12-28:
You actually probably have 2.5.26. Apparently the gphoto2 maintainers did not change the version number in configuration.ac until after the release (https://github.com/gphoto/gphoto2/commit/f2364ea53c53541ba154f049a0998bc461462e18). This means that 2.5.26 will print out version 2.5.23.1. Unfortunately I do not know of any good way to verify which version you have. There is very little difference between these versions of gphoto2, so as long as the library is updated I wouldn't worry about it too much.

Nears, 2020-12-29:
Thanks for your feedback, but I am not sure what you are refering too.
As far as I read the git repo for gphoto2 the rekease 2.5.26 was released at the 11th of october.
[https://github.com/gphoto/gphoto2/compare/v2.5.26...master](url)
And within this release the version tag was increased, so i would expect also gphoto to print the same version.

Or could you tell me were I am wrong? Thanks in advance.

scribblemaniac, 2020-12-29:
The changes you linked are the changes *since* the v2.5.26 release and are not part of the latest stable release. Notice how the version change in AC_INIT is part of these changes.

Nears, 2020-12-29:
Ok, thanks, I tried to use option 1) latest dev instead of 2) last stable release in your script and with this I do get the right version tag. Thanks

closed 2020-12-29 by Nears