gonzalo/gphoto2-updater #82: libgphoto2 not updating to latest version in Raspi OS

koenigd79, 2021-04-07:
I tried the script on an new installation with rasperry pi os on my rpi4.

i choose last stable (2)

but the script end so:
libgphoto 2.5.22 an not 2.5.27

---------------------------------------
--------------------
Finished!! Enjoy it!
--------------------

gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
------------------------

how can i get the lastest version of libgphoto 2.5.27

daniel

rickhewes, 2021-05-05:
Same here...

libgphoto **2.5.22**  not 2.5.27

Any ideas?

scribblemaniac, 2021-05-09:
You both probably have another of libgphoto2 installed from some other source, most likely apt. The script does not remove such packages because of the issues in #57. One thing you can do is run: `ldconfig -p | grep libgphoto2` and share the result here. This will show you the locations where libgphoto2 is installed and accessible. It may be that all you need to do is change the ordering of the paths in your LD_LIBRARY_PATH environment variable or move/symlink some files.

rickhewes, 2021-05-10:
Thanks for the tip.

I was able to run:

```sudo apt-get remove gphoto2 libgphoto2-6 libgphoto2-dev libgphoto2-l10n libgphoto2-port12```

without having to remove any other packages as I'm running a minimal setup. Then the script worked as expected.

Many thanks,

Rick.

gonzalo, 2021-05-14:
thanks @scribblemaniac for keeping an eye on this. I close the Issue

closed 2021-05-14 by gonzalo