gonzalo/gphoto2-updater #83: Finally installed, need to move to main somehow

Athanaetos, 2021-07-01:
Hi, 
I was struggling to get the latest versions, now I have them. However, I can't use any commands as it says "No such file or directory"

I was wondering if I could get some help in sorting that out. 

https://imgur.com/a/9XgzK1q

here is pic of what it returned.

scribblemaniac, 2021-07-01:
Just because you have libgphoto2 (the library) does not mean you have gphoto2 (the command line tool). What was the output of this updater script for you? What is the output of `which gphoto2`? Has the location of gphoto2 moved since it was last executed? If so start a new terminal session or run `hash -d gphoto2` and try again.

Athanaetos, 2021-07-01:
It shows the gphoto2 version in the picture. I did hash -f earlier and it worked. Sorry should have updated my post, I got it all working kind of. So this script has been a huge help. Thanks

closed 2021-07-01 by Athanaetos