gonzalo/gphoto2-updater #84: 404 Error + Sony a6400 support

titaniumschmapple, 2021-12-19:
Hi,

I'm typing in '$ wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh' with Raspberry Pi and getting an 'ERROR 404: Not Found'

I'm trying to get my Sony a6400 camera to be detected with gphoto2 on Raspberry Pi but it's not on the gphoto2 compatibility list (even though a6300 and a6500 are on the list). Does a6400 work or does it need adding?

Thanks

scribblemaniac, 2021-12-19:
> I'm typing in '$ wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh' with Raspberry Pi and getting an 'ERROR 404: Not Found'

At risk of stating the obvious, you should not literally be typing the `$`, that simply indicates the start of the command. As for the 404, it is almost certainly caused by mistyping the url. If you open that url in a browser you should see that it does not lead to a 404 page. If you are still having troubles after checking the url you've typed, you can download the file manually in a browser (you may need to do a "save as", be sure to name it "gphoto2-updater.sh"). Then you can navigate to the directory you saved it to in terminal and run the remainder of the command (`chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh'`)

> I'm trying to get my Sony a6400 camera to be detected with gphoto2 on Raspberry Pi but it's not on the gphoto2 compatibility list (even though a6300 and a6500 are on the list). Does a6400 work or does it need adding?

The Sony a6400 is supported. It is listed on the compatibility page under its model number "Sony ILCE-6400". For future reference, we do not control over what cameras are and are not supported. For that you should go to [the libgphoto repository](https://github.com/gphoto/libgphoto2/issues).

gonzalo, 2022-01-10:
Closed as it's part of libgphoto issues and not for this repository

closed 2022-01-10 by gonzalo