gonzalo/gphoto2-updater #86: LATEST_STABLE_LIBGPHOTO_VERSION undefined

thomaschaaf, 2022-01-10:
This commit breaks the install script as the environment variables are not inlined anymore.

https://github.com/gonzalo/gphoto2-updater/commit/fdb92a9d73e80a0513f1aaccb6095998afa22b0f#diff-630205c4a3115755dbcd9f6e32c96b42fbff83adb31fc2926b8f7198ffc90d28L34

gonzalo, 2022-01-11:
Thanks for reporting, I've extended the readme.md command to also download .env file. This is necessary if we want to keep configuration centralized in one file. Probably we could think about a better way to manage this, but currently is necessary for testing. 
Best option maybe is getting directly the latest stable release version from gphoto2 github repository with some git command. I will explore that option.

scribblemaniac, 2022-01-12:
This was intentional for the reason already described, although I see now there were a couple issues with it. You will need to download the `.env` file along with the script (I was imagining installation being done by cloning the git repository), and additionally you will need to execute the script with the .env file in the current working directory.

I agree that checking for the latest stable release rather than having a hardcoded version would be better and should be possible. When installing that can probably be done relatively easily with git based on the tags. To show that version before the user selects an option there are a few choices:
- The git repository could be downloaded right away. Every option except for quitting needs those repositories, but it still seems strange to be downloading things before the users have taken an affirmative action.
- The latest tag/release could be retrieved from Github. The [releases.atom file](https://github.com/gphoto/libgphoto2/releases.atom) is nicer than the API in that we don't need a key or authentication to retrieve it, but it's harder to parse with bash as an xml file. Maybe we could make grep command that would be good enough to find the latest version number from it without having to fully parse the xml.
- We just don't show the latest stable version number right away.

P.S. @gonzalo In the new instructions, you are missing "wget" before the second url.

gonzalo, 2022-01-13:
You're right @scribblemaniac , I fixed the wget command in the last commit.

I've found a simple script that can provide the latest version tag:
curl --silent "https://api.github.com/repos/gphoto/libgphoto2/releases/latest" | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/'

I will explore how to integrate it with the get allways the last stable release both for testing and running.